Sustaingineering/Pump-Firmware #19: SD Card Overflow Handling

tarek20501, 2020-07-04:


jack5661, 2021-03-21:
Traversing through the Sd Card using openNextFile() is a little weird but its probably starting from oldest file to newest file.

tarek20501, 2021-05-08:
is this the expected output?

```
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

SD Card Type: SDHC
SD Card Size: 7580MB
Total space: 7562MB
Used space: 166MB
Listing directory: /
DIR : /System Volume Information
DIR : /ewaste
FILE: /2020-10-31.txt  SIZE: 2604
DIR : /capstone-mock-vid
FILE: /2021-03-24.txt  SIZE: 1
FILE: /2021-03-25.txt  SIZE: 1
FILE: /2021-03-26.txt  SIZE: 1
FILE: /2021-03-27.txt  SIZE: 1
FILE: /2021-03-28.txt  SIZE: 1
FILE: /2021-03-29.txt  SIZE: 1
FILE: /2021-03-30.txt  SIZE: 1
test\testMain.cpp:23:initSDcard [PASSED]
Writing file: /testSdcard.txt
File written
Reading file: /testSdcard.txt
Read from file: Hello World!
Deleting file: /testSdcard.txt
File deleted
test\testMain.cpp:25:testSdCardWriteRead        [PASSED]
Writing file: /testSdcardAppend.txt
File written
Appending to file: /testSdcardAppend.txt
Message appended
Reading file: /testSdcardAppend.txt
Read from file: HELLLOOO
GOODBYE
Deleting file: /testSdcardAppend.txt
File deleted
test\testMain.cpp:26:testSdCardAppendFile       [PASSED]
Writing file: /testSdcardDelete.txt
File written
Deleting file: /testSdcardDelete.txt
File deleted
test\testMain.cpp:27:testSdCardDeleteFile       [PASSED]
Writing file: /2021-03-21.txt
File written
Writing file: /2021-03-22.txt
File written
Writing file: /2021-03-23.txt
File written
Writing file: /2021-03-24.txt
File written
Writing file: /2021-03-25.txt
File written
Writing file: /2021-03-26.txt
File written
Writing file: /2021-03-27.txt
File written
Writing file: /2021-03-28.txt
File written
Writing file: /2021-03-29.txt
File written
Writing file: /2021-03-30.txt
File written
Listing directory: /
DIR : /System Volume Information
DIR : /ewaste
FILE: /2020-10-31.txt  SIZE: 2604
FILE: /2021-03-21.txt  SIZE: 1
DIR : /capstone-mock-vid
FILE: /2021-03-22.txt  SIZE: 1
FILE: /2021-03-23.txt  SIZE: 1
FILE: /2021-03-24.txt  SIZE: 1
FILE: /2021-03-25.txt  SIZE: 1
FILE: /2021-03-26.txt  SIZE: 1
FILE: /2021-03-27.txt  SIZE: 1
FILE: /2021-03-28.txt  SIZE: 1
FILE: /2021-03-29.txt  SIZE: 1
FILE: /2021-03-30.txt  SIZE: 1
Skipping /System Volume Information
Skipping /ewaste
Deleting file: /2020-10-31.txt
File deleted
Deleting file: /2021-03-21.txt
File deleted
Skipping /capstone-mock-vid
Deleting file: /2021-03-22.txt
File deleted
Listing directory: /
DIR : /System Volume Information
DIR : /ewaste
DIR : /capstone-mock-vid
FILE: /2021-03-23.txt  SIZE: 1
FILE: /2021-03-24.txt  SIZE: 1
FILE: /2021-03-25.txt  SIZE: 1
FILE: /2021-03-26.txt  SIZE: 1
FILE: /2021-03-27.txt  SIZE: 1
FILE: /2021-03-28.txt  SIZE: 1
FILE: /2021-03-29.txt  SIZE: 1
FILE: /2021-03-30.txt  SIZE: 1
Reading file: /2021-03-21.txt
[E][vfs_api.cpp:64] open(): /sd/2021-03-21.txt does not exist
Failed to open file for reading
Reading file: /2021-03-22.txt
[E][vfs_api.cpp:64] open(): /sd/2021-03-22.txt does not exist
Failed to open file for reading
Reading file: /2021-03-23.txt
Read from file: 0
test\testMain.cpp:91:testSdCardOverflow:FAIL: Expected NULL     [FAILED]
-----------------------
5 Tests 1 Failures 0 Ignored
====================================================================================================== [FAILED] Took 27.99 seconds ======================================================================================================
Test    Environment          Status    Duration
------  -------------------  --------  ------------
*       esp32doit-devkit-v1  FAILED    00:00:27.987
================================================================================================= 1 failed, 0 succeeded in 00:00:27.987 =================================================================================================The terminal process "C:\Users\tarek\.platformio\penv\Scripts\platformio.exe 'test', '--environment', 'esp32doit-devkit-v1'" terminated with exit code: 1.

```

tarek20501, 2021-05-08:
@jack5661 it seems like it did the right thing but it is the test that returned a false negative. Do you agree??

jack5661, 2021-05-09:
@tarek20501 it does look like it did the right thing. The behaviour for that test case is that it will delete the oldest 3 files then check the remaining files. I didn't take into account the SD card may have files created from before running the test which is why it deleted the "/2020-10-31.txt" file.

PR #127 (2021-05-15): #19 sd card ovf handling
closes #19
closed 2021-05-15 by tarek20501