Sustaingineering/Pump-Firmware #34: Initial Voltage Sensor Support

tarek20501, 2020-08-31:


tarek20501, 2020-09-03:
Done as of c09bd5dfa14ac39840ba84ce9e5e48768ae7c05a waiting for hardware to test.

tarek20501, 2020-09-08:
Need to validate equations and check the following from the datasheet
* ADC Resolution
* Max Voltage for analog pins

tarek20501, 2020-10-15:
https://www.mouser.com/datasheet/2/389/cd00237391-1796869.pdf

tarek20501, 2020-10-24:
- Check ADC resolution of Particle
- Ask Forbes how he got the 3.9 from ESP32 datasheet
- Do Step 2 for Particle

RyanL17, 2020-10-24:
STM32F205 Family: 12-bit, 10-bit, 8-bit or 6-bit configurable resolution
Particle 2nd Gen Boards: 12-bit
I believe 4095 is used because it starts at 0 (4096 ints) 
3.6V max analog reference voltage

STM32F205 Family Reference Manual:
https://www.st.com/content/ccc/resource/technical/document/reference_manual/51/f7/f3/06/cd/b6/46/ec/CD00225773.pdf/files/CD00225773.pdf/jcr:content/translations/en.CD00225773.pdf

Particle Analog Read Reference:
https://docs.particle.io/reference/device-os/firmware/photon/#analogread-adc-

closed 2020-10-24 by RyanL17