Sustaingineering/Pump-Firmware #64: Automate code formatting

tarek20501, 2020-10-10:


ShreyansK2000, 2020-10-10:
We can add this below the general section as it is (please try it out first, the settings.json would be shared across the repo):

### Visualizing the max character guideline

For item 1: Make sure that your workspace's settings.json file has the following rule: **`"editor.rulers": [100]`**. You can add more vertical rulers at different character limits for tighter code in the array if you like

### Formatting
For items 2, 3, and 4: Make sure that you have installed the **`C/C++`**  IntelliSense, debugging, and code browsing extension by Microsoft. Also make sure that your Workspace's default indentation configuration is 4 spaces (**`Ctrl + Shift + P`**->"**Indent Using Spaces**"->**4**).

 - From inside a C++ file, press **`Ctrl + Shift + P`** to open the command pallette. Alternatively, press **`Ctrl + P`** and type "**>**" before the next step
 - Search for and select the "**Format Document With**" option. You should see "**C/C++**" listed along with any other formatting extensions you may have installed. Find and select "**Configure Default Formatter...**" at the end of the list
 - Once again,  a list of valid formatting extensions will appear. Select "**C/C++**". At this point pressing **`Shift + Alt + F`** at any point will reformat the code to the style discussed above

#### Automating the formatting

Once you have completed the above steps, you may optionally choose to enable formatting on save or formatting as you type. To do this, in the settings.json file of your workspace add:

**`"editor.formatOnSave": true`**: This will automatically format the file each time you press `Ctrl + S` or save manually.

**_OR_**

**`"editor.formatOnType": true`**: This will automatically format the file each time you reach the end of a line, i.e. a semicolon **`;`** character.

jack5661, 2020-10-24:
Found plugin that uses the C/C++ plugin formatting to format all files at once. The plugin is called "Format Files" by jbockle.

Recommend to use command "Start Format Files: From Glob" to format select files. (Press F1 to open command palette)

closed 2020-11-28 by tarek20501