Sustaingineering/Pump-Firmware #66: Figure out the frequency of the publishing

tarek20501, 2020-10-17:
Experiment with `Cellular.getDataUsage()` to figure out the frequency of publishing such that we do not exceed 3MB/month

ShreyansK2000, 2020-10-24:
Using the `Cellular.resetUsage()` method to set all internal counters to 0 at every power cycle, we were able to test for the size of packets sent and received for every publish with the use of webhooks. The test conducted published 2 times each, with 10 cycles of the sampling in between them to allow for stabilization of the values. For now, we conducted 10 trials.

### Trial 1, 7

|         	| TX  	| RX  	| Data(i)<sub>TX</sub> - Data(i-1)<sub>TX</sub>	| Data(i)<sub>RX</sub> - Data(i-1)<sub>RX</sub> 	|
|---------	|-----	|-----	|---------	|-----------------------------------------------	|
| **Data(0)** 	| 287 	| 287 	|         	|                                               	|
| **Data(1)** 	| 394 	| 348 	| 107     	| 61                                            	|
| **Data(2)** 	| 501 	| 409 	| 107     	| 61                                            	|

### Trial 2-6, 8-10

|         	| TX  	| RX  	| Data(i)<sub>TX</sub> - Data(i-1)<sub>TX</sub>	| Data(i)<sub>RX</sub> - Data(i-1)<sub>RX</sub> 	|
|---------	|-----	|-----	|---------	|-----------------------------------------------	|
| **Data(0)** 	| 598	| 414 	|         	|                                               	|
| **Data(1)** 	| 705 	| 475 	| 107     	| 61                                            	|
| **Data(2)** 	| 812 	| 536 	| 107     	| 61                                            	|

In all cases, regardless of the initial data consumed for the handshake, the total data exchanged for a publish event is 107 + 61 = **168 bytes**. We can leverage this to calculate the frequency of `publish` events

tarek20501, 2020-10-24:
Next week let's try getting data usage at the beginning of the setup and reset. See if you get consistent data usage for the cloud handshake

tarek20501, 2020-10-24:
Let `Data(n)` be the function that returns data usage at nth `Publish` call where `n = 0` represent data usage resulted from the initial handshake with the Particle Cloud.
The following is the algorithm that calculate publishing period:
```
Data(2) - Data(1) = MessageSize
MonthlyDataLimit / MessageSize = NumberOfMessages/Month

NumberOfMessages/Month * Month/31 = NumberOfMessages/Day

NumberOfMessages/Day * Day/24 = NumberOfMessages/hour

NumberOfMessages/hour * hour/(60*60) = NumberOfMessages/second

second/NumberOfMessages = PeriodOfPublishing (seconds)
```

ShreyansK2000, 2020-10-31:
I conducted a few trials this week to see how many bytes are sent and received over the initial handshake. The following are the results:

| Trial |  TX  |  RX  | Data(0) |
|:-----:|:----:|:----:|--------|
|   1   | 3109 | 2365 | 5474   |
|   2   | 3111 | 2361 | 5472   |
|   3   | 3111 | 2365 | 5476   |
|   4   | 3109 | 2363 | 5472   |
|   5   | 3109 | 2363 | 5472   |
|   6   | 3111 | 2361 | 5472   |
|   7   | 2704 | 1783 | 4487   |
|   8   | 3111 | 2363 | 5474   |
|   9   | 3111 | 2365 | 5476   |
|   10  | 3511 | 2486 | 5997   |
|   11  | 3107 | 2361 | 5468   |
|   12  | 3109 | 2365 | 5474   |
|   13  | 3111 | 2365 | 5476   |
|   14  | 3111 | 2365 | 5476   |
|   15  | 3107 | 2365 | 5472   |

Given the larger than normal TX and RX values in trial 10, I think we should consider something around 6200 bytes or higher as the handshake establishment bytes in our calculation for total bytes used.

tarek20501, 2020-10-31:
Thanks for your dedication @ShreyansK2000 
Next time we meet up will turn the algorithm into code.
Just one question...
can you tell me where you `resetUsage` and `getUsage` calls for your last experiment?

ShreyansK2000, 2020-11-01:
`getUsage` at the start of `setup()`, then for 10 calls of `loop(`), I let the numbers stabilize, and then call `resetUsage`. I also let that run for at least 10 loops each time just in case, after reset TX and RX values always stayed at 0.

closed 2020-11-28 by tarek20501