Sustaingineering/Pump-Firmware #74: Revise LED Pattern

tarek20501, 2020-11-14:
We need to  determine a pattern for the LED on Particle and ESP32 to let the user know that the system is initialized and operating nominally

jack5661, 2021-01-30:
**LED Patterns:**

**Initializing** - Quickly blinking LED
**Initialized** - Let LED stay solid for 1 minute to signify everything is okay then turn off to save power
**Initializing Failed** - Turn LED off for 10 seconds then blink LED every 5 seconds 
**Low Battery/File Space** - Blink LED every 10 seconds

jack5661, 2021-01-30:
Need infrastructure or way to tell if setup has failed.

RyanL17, 2021-03-27:
New LED Pattern:

LED OFF: Initialization Complete
Flashing While Initializing: On when entering initialization function, Off when back to main
LED ON: Initialization Failed

closed 2021-05-15 by RyanL17