Sustaingineering/Pump-Firmware #83: Add current sensor (HSTS016L) support

tarek20501, 2021-01-23:
Add initial support for new current sensor

tarek20501, 2021-07-18:
@forbes-choy 
Please comment the new equation that converts the analog output to current in Amps

tarek20501, 2021-07-25:
https://cdn.shopifycdn.net/s/files/1/2723/6858/files/HSTS016L_a7e0ee3a-45cc-4504-95cc-f8ae5a5ae776.pdf?v=1614127414
[image]

`V = 2.5 + (I/I_PN) * 0.625` where `I_PN` is rated input
`I = (V - 2.5) * (I_PN/0.625)`

tarek20501, 2021-07-25:
@forbes-choy Let me know if I got this correctly...

```
#define I_PN 10 // can be 10,  20, 30, 50, 100, 150, or 200
float Current::readRaw()
{
    int analogData = analogRead(m_pin); // Take reading
    // Convert to decimal
    float hallVoltage = analogData * (m_maximumVoltage / analogDigitalConversionResolution);
    // Compute the current from the voltage reading equation
    float hallAmps = (hallVoltage - 2.5) * (I_PN / 0.625);
    return hallAmps;
}
```

forbes-choy, 2021-07-27:
@tarek20501 your Current::readRaw() function looks good to me. We will test the hardware ASAP

closed 2021-08-07 by tarek20501