Sustaingineering/Pump-Firmware #121: Experiment with the free plan

tarek20501, 2021-03-20:


EliBCC, 2021-03-27:
## Free plan restrictions

- Up to 100 devices
- Up to 100K Data Operations* per month
- Up to 100 MB cellular data per month

\* Data operations include every publish, subscribe, function, or variable. There is no lower limit on how many bytes counts as a data operation, but each operation must be less than 622 bytes. Firmware updates, internal events, acknowledgments, webhooks, and cloud API calls do not count as data operations

## How to track usage
The only way to track usage is through console.particle.io. Usage cannot be tracked locally on the Particle

## Exceeding the limit
If one of the three limits are exceeded, data operations will not complete until the end of the month

## Reducing data usage
- Combine publish events using csv, json, etc.
- Aggregate by time
- Only transmit on change

## Sources
https://docs.particle.io/tutorials/device-cloud/introduction/#minimizing-data-operations
https://support.particle.io/hc/en-us/articles/360039741073-Pricing-FAQ
https://blog.particle.io/our-new-pricing-model/

tarek20501, 2021-03-27:
Please add a `numberOfDevices`  member for now

EliBCC, 2021-04-03:
# Specifications

1. No more than 10k publish events per account should be sent per month
2. For simplification, no less than (31days * 24hours * 60minutes * 60seconds)/10k messages should elapse between publish
 a. When persistent variables are implemented, this may adapt intelligently
3. Each published message should be no more than 622 bytes when received
 b. To ensure this, messages should be capped at 600 bytes when sent

## Packet format

`11.11, 11.11, 11.11, 11.11, 11.11, 11111,; 11.11, 11.11, 11.11, 11.11, 11.11, 11111,; 11.11, 11.11, 11.11, 11.11, 11.11, 11111,; 11.11, 11.11, 11.11, 11.11, 11.11, 11111,; 11.11, 11.11, 11.11, 11.11, 11.11, 11111,; `

Description: Each message consists of 5 floats (sensor values) and one int (utime), each of variable length. Values are delimited by commas. Packets consist of no more than 600 bytes, which may contain any number of messages. Messages are delimited by semi-colons.

## Questions
Should the data in the buffer (about 15 messages) be the most recent data, or a representative sample across the period?

## Future specs
1. Allow for both free and paid plans
2. Implement persistent variables to keep track of total messages sent this month
3. Implement a webhook to track official number of operations
4. Make number of devices controlled from Particle console

ShreyansK2000, 2021-05-15:
### GSMTests.cpp (based on free plan implementation)
```cpp
#include "GSMTests.h"
#if EN_GSM == 1

String message1 = "313.31,314.31,315.31,316.31,317.31,1617476161";
String message2 = "312.31,314.31,315.31,316.31,317.31,1617476162";
String message3 = "314.31,314.31,315.31,316.31,317.31,1617476163";
String message4 = "3111.31,314.31,315.31,316.31,317.31,1617476164";
String message5 = "3166.31,314.31,315.31,316.31,317.31,1617476165";
String message6 = "0.31,314.31,315.31,316.31,317.31,1617476166";
String message7 = "1.31,314.31,315.31,316.31,317.31,1617476167";

void testPublishInterval()
{
    String testTag = "PUBLISH INTERVAL";
    printTestStart(testTag);
    bool isPublished = false;
    
    int lastPublishTime;
    int publishCount = 0;
    while(!isPublished) {
        delay(800);
        isPublished = gsm.Publish(String(0), message1);
        publishCount++;
        Serial.printf("Published %d times so far (loop 1)\n", publishCount);
    }

    Serial.println("Exited loop 1, updating lastPublishTime variable");
    lastPublishTime = rtc.getTimeStamp().toInt();

    isPublished = false;
    int newPublishTime;
    publishCount = 0;
    while(!isPublished) {
        delay(800);
        isPublished = gsm.Publish(String(0), message2);
        publishCount++;
        Serial.printf("Published %d times so far (loop 2)\n", publishCount);
    }

    Serial.println("Exited loop 2, updating newPublishTime variable");
    newPublishTime = rtc.getTimeStamp().toInt();

    Serial.println("Checking for difference in recorded times");
    if ((double)(newPublishTime - lastPublishTime) < (31.0 * 24.0 * 60.0 * 60.0)/(10000.0)) {
        Serial.println("Publish Interval Test FAILED");
        // return false;
    } else {
        Serial.println("Publish Interval Test PASSED");
        // return true;
    }
   printTestEnd(testTag);
}

void testPublishLength()
{
    return;
}

void testGsm(){
    void testPublishInterval();
}

#endif
```

### GSMTests.h
```cpp
```

jack5661, 2021-05-15:
max operations per day: 100,000 operations / 31 days / 10 particles = 322.5806
Each operation: 582 bytes + header (40 bytes for TCP header) = 622 bytes
Total bytes sent: 322 * 582 = 187,404 
Total seconds in day: 86,400
Want to send every: 86,400 / 322 = 268.323  
Message/second = < 60 bytes (Overestimate)
Each data operation will be: ~540 bytes + header
9 messages / 270

- Send every 30 data point into buffer 
	- Send about ~1.31 MB of data/day
	- Could add feature to Web App to analyze log files
- Still want to log every data point into SD card files
- Use "\n" as delimiter bc we are adding it in main.cpp anyways

Implementation:
- Count number of messages in buffer
- Add every 30th data to buffer 
- Add initialization method to compute all the math at the start (frequency of messages)
- Use "\n" as delimiter bc we are already using it

PR #138 (2021-06-19): #121 gsm particle
Addresses Free Plan changes by limiting the number of particle cloud publishes issues. This depends on the total number of particles. Closes #121 and #115 (some minor adjustments were made to the testing framework as well).
closed 2021-06-19 by tarek20501